open-ephys/ephys-test-board #12: Alleviate the Squish of PCB Between Aluminum Enclosure

cjsha, 2024-12-19:
Perhaps the best way to do this is with an elastic washer

jonnew, 2025-01-05:
I think that there might be an opportunity for grounding here (#11). 

- Put paste layer on the top/bottom of the corner holes, so that there is soft solder layer on both sides
- Add some dimpling/spike features on the top and bottom around the holes.
- Sand these dimples before installation to remove anodization. 
- Screw together so that the dimples smash into solder ring

cjsha, 2025-01-06:
That could work. Also, browse this [Digikey page](https://www.digikey.com/en/products/filter/rfi-and-emi-contacts-fingerstock-and-gaskets/945?s=N4IgjCBcpgnAHLKoDGUBmBDANgZwKYA0IA9lANogAMIAugL7EC0ATMiGpAC4BOArkVIUQAVhDEkDRiABs7AJYATKCCZgqEYgAcuK8SC4BPLfhWZcaevSA) for additional ideas.

PR #38 (2025-04-03): Address issues found during prototyping Rev. B
- Add recess for reset button
- Add 3.5V power management circuit and simplify on/off switch debounce
- Increase space in bottom portion of case for thicker battery
- Remove alignment notches from PCB -- uses features of case instead
- Fixes #37 
- Fixes #36 
- Fixes #12 
- Fixes #35
- Fixed #34
closed 2025-04-03 by jonnew