open-ephys/ephys-test-board #34: Different battery dimension for mass production

filcarv, 2025-03-22:
The battery used is Adafruit's [Product ID: 258 ](https://www.adafruit.com/product/258) which has these dimensions: **34mm** x **62mm** x **5mm**. This battery is out of stock.

We found a reliable and deterministic distributor for the Akyga Battery model LP623255, which has slightly different dimensions (**33mm** x **57.5mm** x **6.7mm**). Can we use this battery?

jonnew, 2025-03-26:
Probably. I will need to make the bottom cavity deeper.

jonnew, 2025-03-29:
This battery is fine in terms of its electrical characteristics, but the connector it uses is wire to wire only, as far as I can tell. That means you would need to solder some wire component into the board. I would much prefer using a battery with a connector that can mate to a connector on the board (like the current one). Can you please find another option?

jonnew, 2025-04-03:
Its fine, thanks for the insight. If i was to design again IO might put a cutout for one of these, but they are too large for the current design.

I think the solution is to;

1. Leave the current connection
2. Put some direct solder pads next to it for a direct soldered connection or to put a mating cable on.

PR #38 (2025-04-03): Address issues found during prototyping Rev. B
- Add recess for reset button
- Add 3.5V power management circuit and simplify on/off switch debounce
- Increase space in bottom portion of case for thicker battery
- Remove alignment notches from PCB -- uses features of case instead
- Fixes #37 
- Fixes #36 
- Fixes #12 
- Fixes #35
- Fixed #34