open-ephys/ephys-test-board #35: Recessed space for label in the case

filcarv, 2025-03-22:
Open Ephys products use a label of 30mm x 50mm with the Open Ephys logo, the SKU, and some legally required information.

Is it possible to add a recessed space somewhere in the case 0.5mm deep and 31mm x 51mm in size? This will be used to add the label.

jonnew, 2025-03-26:
yes should be doable. We should also put some labels on the case, (e.g. on/off), signal input etc. How do you want to do that?

jonnew, 2025-04-01:
A sticker with manufacturing information will go in a recessed position on the bottom of the device. The top should have a printed layer (or laser engraved if the case is made from milled aluminum) with something like the following:

[image]

- The 30x50 mm recess is now added to the bottom of the case. 
- The above design is included in top-silk.pdf.

PR #38 (2025-04-03): Address issues found during prototyping Rev. B
- Add recess for reset button
- Add 3.5V power management circuit and simplify on/off switch debounce
- Increase space in bottom portion of case for thicker battery
- Remove alignment notches from PCB -- uses features of case instead
- Fixes #37 
- Fixes #36 
- Fixes #12 
- Fixes #35
- Fixed #34
closed 2025-04-03 by jonnew